evil-mad/EggBot #147: Hatch fill not working in Inkscape v1.1-dev (78e9eb4, 2020-12-04)

DavidGP, 2020-12-05:
Here is the error window that I get:
[image]

The text is as follows:

`eggbot_hatch.py:621: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:625: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:629: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:633: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:637: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:641: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:645: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:649: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option(
eggbot_hatch.py:653: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--tab",  # NOTE: value is not used.
eggbot_hatch.py:1736: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
eggbot_hatch.py:686: DeprecationWarning: simpletransform.parseTransform -> Transform(str).matrix
  self.docTransform = simpletransform.parseTransform('scale({0:f},{1:f})'.format(sx, sy))
eggbot_hatch.py:1215: DeprecationWarning: Effect.selected is now a dict in the SvgDocumentElement class. Use `self.svg.selected`.
  self.recursivelyTraverseSvg([self.selected[id_]], self.docTransform)
eggbot_hatch.py:818: DeprecationWarning: simpletransform.parseTransform -> Transform(str).matrix
  mat_new = simpletransform.composeTransform(mat_current, simpletransform.parseTransform(node.get("transform")))
eggbot_hatch.py:818: DeprecationWarning: simpletransform.composeTransform -> Transform(M1) * Transform(M2)
  mat_new = simpletransform.composeTransform(mat_current, simpletransform.parseTransform(node.get("transform")))
eggbot_hatch.py:894: DeprecationWarning: simplepath.formatPath -> str(element.path) or str(Path(array))
  self.addPathVertices(simplepath.formatPath(a), node, mat_new)
Traceback (most recent call last):
  File "eggbot_hatch.py", line 1736, in <module>
    e.affect()
  File "F:\Portable Programme\inkscape\share\inkscape\extensions\inkex\deprecated.py", line 182, in affect
    return self.run(args=args)
  File "F:\Portable Programme\inkscape\share\inkscape\extensions\inkex\base.py", line 141, in run
    self.save_raw(self.effect())
  File "eggbot_hatch.py", line 1215, in effect
    self.recursivelyTraverseSvg([self.selected[id_]], self.docTransform)
  File "eggbot_hatch.py", line 894, in recursivelyTraverseSvg
    self.addPathVertices(simplepath.formatPath(a), node, mat_new)
  File "F:\Portable Programme\inkscape\share\inkscape\extensions\inkex\deprecated.py", line 259, in _inner
    return func(*args, **kwargs)
  File "F:\Portable Programme\inkscape\share\inkscape\extensions\inkex\deprecated-simple\simplepath.py", line 36, in formatPath
    return str(Path(a))
  File "F:\Portable Programme\inkscape\share\inkscape\extensions\inkex\paths.py", line 1128, in __init__
    self.append(PathCommand.letter_to_class(item[0])(*item[1]))
  File "F:\Portable Programme\inkscape\share\inkscape\extensions\inkex\paths.py", line 116, in letter_to_class
    return PathCommand._letter_to_class[letter]
KeyError: 'M '
`

oskay, 2020-12-05:
First, we do not have any release that supports Inkscape 1.1 yet. We do plan to offer support for it _once it is released_. If you have trouble running this software in the release version of Inkscape 1.1, please feel free to open an issue for that.

Second, it looks like you're running the Inkscape 0.9x version of the EggBot software.  I do not expect it to work under Inkscape 1.0, let alone 1.1.

closed 2020-12-05 by oskay

DavidGP, 2020-12-05:
Thanks for the quick response. 

Sorry that I am a little lost here. Which versions of Inkscape and EggBot should I use in order to be able to create vector hatches?

oskay, 2020-12-05:
If you're using Inkscape 0.9, this version should work for you. If you're using Inkscape 1.0+, use the current versions which are now bundled with the AxiDraw installers: https://wiki.evilmadscientist.com/Installing_software

DavidGP, 2020-12-05:
Thanks very much, the manual install worked:
https://wiki.evilmadscientist.com/Installing_software#Alternative_.28Manual.29_install