evil-mad/EggBot #148: Windows EBB updater: Use EBB in bootloader mode?

oskay, 2020-12-09:
If an EBB is in bootloader mode, it will not currently be detected and updated by the updater program. It would be ideal to catch and handle this case.

EmbeddedMan, 2020-12-09:
That's a fantastic idea. It won't be straight forward to implement though. Once an EBB is in bootloader mode, it has already disconnected itself as a CDC ACM (virtual serial port) and reattached itself as  HID device. So we'll need a way for the updater to query the HID devices on the computer and see if any match the EBB in bootloader mode.

The other option might be to add a feature to the command line HID bootloader software to try and detect if there are any available EBB boards to update, and then return a result code that the updater app could use to try and detect an EBB in bootloader mode.

oskay, 2020-12-09:
Yeah; it's not quite trivial. 

We have a feature in the Mac updater that does this, but crudely. It checks for available EBBs (in the usual way), but if it doesn't find any, guesses if a port name is *plausible* for having an EBB and tries running the mphidflash program.

EmbeddedMan, 2023-09-30:
Oh yeah! We already have this as a previous issue. OK, well, #217 covers us for this one too then. Nice.

EmbeddedMan, 2024-05-23:
This feature is now in the v3.0.2 Windows Updater that got merged into master as of PR #232 .

closed 2024-05-23 by EmbeddedMan