evil-mad/EggBot #150: Linux Inkscape error on apply

evanRubinsteinIT, 2021-04-03:
Upon launching the extension in inkscape and hitting the apply button on any of the menu options, I get the following error:

```
Traceback (most recent call last):
  File "eggbot.py", line 41, in <module>
    plot_utils = from_dependency_import('plotink.plot_utils')  # Requires v 0.15 in plotink
  File "/home/evan/.config/inkscape/extensions/plot_utils_import.py", line 36, in from_dependency_import
    module = import_module(module_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/evan/.config/inkscape/extensions/axidraw_deps/plotink/plot_utils.py", line 40, in <module>
    from math import sqrt, isclose
ImportError: cannot import name isclose
``

Catfang, 2021-04-03:
Same issue. I'll update if I make any progress.

oskay, 2021-04-03:
This software requires python 3.5+ in the current version.

oskay, 2021-04-03:
> I have the same issue except that I'm on Windows.

If you're getting this issue on Windows, it suggests that you're using Inkscape < 1.0; update to the current version, Inkscape 1.0.2

closed 2021-04-03 by oskay