evil-mad/EggBot #151: Add "QE" (Query motor Enables and resolution) to EBB firmware

EmbeddedMan, 2021-04-22:
It is possible to read the I/O pins directly to see if either driver is enabled, and what the common MS1/MS2/MS3 (resolution) bits are set to. However, it would be much more convenient if there were just one command to read this status information out and return it.

No need to put it into the motion queue - it should execute immediately.

Maybe return the form
<Motor1Enable>,<Motor1Resolution>,<Motor2Enable>,<Motor2Resolution><CR>OK<CR>
where the enables are either 0 (disabled) or 1 (enabled), and the resolution is 
0 for full step
1 for 1/2 step
2 for 1/4 step
3 for 1/8th step
4 for 1/16th step

For EBB, both motors will always have the same resolution (since they can't be set separately with the current EBB hardware).

oskay, 2021-04-22:
Another approach would be to follow the same format as EM:

`<Status_1>,< Status_2><CR>OK<CR>`,

where

       If Status_i == 0, -> Motor disabled
        If Status_i == 1, -> 16X microstepping
        If Status_i == 2, -> 8X microstepping
        If Status_i == 3, -> 4X microstepping
        If Status_i == 4, -> 2X microstepping
        If Status_i == 5, -> No microstepping

This is unfortunately limiting, should higher microstep values be available in the future.

EmbeddedMan, 2021-04-22:
What if we just reversed the list:
Status_i = 0 -> motor disabled
Status_i = 1 -> full step
Status_i = 2 -> 1/2 step,
etc.

Then we could make the value as high as we needed to go.

oskay, 2021-04-22:
Sounds good.

0 -> Motor disabled

Otherwise, Motor enabled with step size 1 / 2^(n-1).

Though, I wonder: Is there any reason not to just return reciprocal step size directly, rather than as a power of 2?

What about:

0 -> Motor disabled
1 -> Full steps
2 -> 1/2 steps
4 -> 1/4 steps
8 -> 1/8 steps
16 -> 1/16 steps?

EmbeddedMan, 2021-04-22:
No reason. It's just a look up table in the firmware anyway. We'll go with that.

EmbeddedMan, 2021-04-28:
Fixed/added in the EBB v2.8.0 firmware release.

closed 2021-04-28 by EmbeddedMan