evil-mad/EggBot #159: Inconsistent line endings

mutoo, 2021-06-03:
I am working on a side project using web-usb to communicate with my AxiDraw. I started this by reading the [EBB Command set documentation](http://evil-mad.github.io/EggBot/ebb.html). It was great but I found that the **Line Endings** for each command are really inconsistent. It's looking for `<CR><NL>`, sometimes `<NL><CR>`.

I also found that some Response in the doc saying the line ending is `<NL><CR>`, but I actually got `<CR><NL>` from the serial port. e.g. [LM](http://evil-mad.github.io/EggBot/ebb.html#LM), [LT](http://evil-mad.github.io/EggBot/ebb.html#LT).

Then I did some investigation on the code base, and I realized there is actually only one command using `<NL><CR>` as line ending: [QM](http://evil-mad.github.io/EggBot/ebb.html#QM). 

I am not sure fixing those line endings will cause a breaking change. So what I can do for now is just fixing the inconsistency in the documentation.

oskay, 2021-06-03:
Thank you; this is an improvement.

We are a little bit embarrassed about the way these line endings have evolved. If a firmware rewrite update should ever present an opportunity for a full rewrite with breaking changes, this is (and has been for some time) near the top of our list.

EmbeddedMan, 2021-06-04:
The embarrassment is fully on my shoulders. The fact that EBB has inconsistent line endings is a stupid oversight on my part, that, had I corrected it quickly, never would have been a big deal. But now there may be folks that depend on those couple commands that don't follow the pattern, so we can't change it in the code.

Your fix of the documentation is a good way to handle the problem. Thank you for submitting this PR.

I'm going to close the issue now as the change has been merged.

closed 2021-06-04 by EmbeddedMan