evil-mad/EggBot #169: Boot up default for servo on B0 changed from firmware version 2.56 to 2.6+

cyclicredundancy, 2021-12-27:
I am using EBB with watercolor bot 1.0 (EBB board version 2.3) and while updating the EBB firmware noticed that the servo on B0 now boots up to a different pen position, pen up vs. pen down previously. And somehow on my board this causes my servo to continuously make a clicking sound. Not sure if the clicking sound is because of an issue with my servo or something in the firmware.

oskay, 2021-12-27:
The power on position is nominally "pen up", and has been (at least) from 2.5.5 to present. However, this is only a default position, not the actual machine-specific nor software-specific pen-up/pen-down positions. The exact setpoints for the power on pen-up and pen-down position did change in 2.6.0 (2018).

Part of the WCB 1.0 assembly process is calibrating the servo positions. If you're changing servo motors, changing software, or changing firmware, you likely need to calibrate the servo position again-- no two servos work exactly the same, and the different software and firmware versions have different setpoints as well.

As for your servo, it may be end of life, or it may be trying to move out of range.

cyclicredundancy, 2022-01-15:
Thanks! Closing this as it is not related to EBB firmware.

closed 2022-01-15 by cyclicredundancy