evil-mad/EggBot #171: Extract hatching to a more flexible function?

migurski, 2022-01-11:
Would you be receptive to making the hatching functionality easier to use directly, separate from InkScape or SVG document trees? I currently use it to hatch whole documents, but in many cases I need finer control. For example, I might know the thickness of a specific pen and need to hatch so it fills evenly with ink. Or I have a set of different shapes which would benefit from per-shape hatching orientation within a single document.

I’ve looked at the code in [`inkscape_driver/eggbot_hatch.py`](https://github.com/evil-mad/EggBot/blob/master/inkscape_driver/eggbot_hatch.py) and it feels very tied to InkScape’s internal representation of SVG documents. 

Is there a place for a more portable hatching implementation in this project?

oskay, 2022-01-11:
We have separate work underway adding clipping that will (once merged) permit a different approach to hatching, so we're not planning any major changes to the existing codebase. 

The Hatch Fill extension does work on *selected* objects, or all objects if none are selected. If your objects are separated by layer or style such that they can be selected that way, it may be a head start. You can also potentially run the script with a programmatic selection.

migurski, 2022-01-11:
I’ll see what I can do programmatically. I run everything via `axicli` and try to never have to open InkScape. I hope the new approach will be compatible!

closed 2022-01-11 by migurski