evil-mad/EggBot #172: Test out shutting off servo control pulse when powering servo down to increase servo life

EmbeddedMan, 2022-02-22:


EmbeddedMan, 2022-05-19:
A potential fix for this is in the Issue_172 branch. More testing must be done before it should be considered for merging. This change makes the servo I/O pin an output driven low any time the power to the serovo is shut off.

EmbeddedMan, 2023-09-22:
OK, well, the code for this is on Issue_172 branch, and has not been folded into the v3.0.0 branch. Unless there's interest in moving this forward I propose closing it.

EmbeddedMan, 2024-05-23:
@oskay While we do have working code for this feature on a branch, it feels like the standardization of the much more robust pen lift motors with new machines means this feature is no longer needed. OK to close this issue and delete the associated branch, or should we keep this around in case we want to fold it into a future firmware release?

oskay, 2024-05-23:
Let's close it. We're just not planning to invest more resources in low-end servo motor performance.

closed 2024-05-23 by oskay