evil-mad/EggBot #174: EBB Updater fails under Windows 11 due to not being able connect to EBB in bootloader mode

EmbeddedMan, 2022-04-16:
Attempted to use the EBBUpgraderGUI_v280.exe under Windows 11 to bring an EBB up from 2.7.0 to 2.8.0. Updater failed with : 

[image]

There might be something different between Win11 and previous versions that has to do with support for the HID based bootloader protocol from Microchip (used for bootloading new firmware to EEBs).

After the error, the updater leaves the EBB in bootloader mode (alternating red/green LEDs). A possible workaround for this problem might be issue #148 (to allow programming EBBs that are already in bootloader mode).

EmbeddedMan, 2022-04-16:
Further information : the Windows 11 system on which this happened is a pretty clean system, without any EBB or UBW drivers on it. I had just installed AxiDraw_320.exe to get the EggBot drivers onto the system.

oskay, 2022-04-16:
Interesting; I did hear of this error from someone recently-- It did not occur to me to ask if this was Win 11.

EmbeddedMan, 2022-04-16:
Additional information: When I downloaded the Microchip provided bootloader application (
https://www.schmalzhaus.com/EBB/EBB_BL/HIDBootLoader.zip) and then used it to load the EBF_v2.8.0.hex file from this GH repo, it worked just fine. (I.e. doing the update 'manually')

So my current guess is that the mphidflash program has some difficulty with Windows 11 and seeing the board in bootloader mode, but the Microchip bootloader application (Microchip USB HID Bootloader v2.6a) does not.

EmbeddedMan, 2022-05-20:
Further testing showed that, at least in my case, the problem was not really with mphidflash. Rather, Windows 11 takes slightly longer to re-enumerate the EBB once in bootloader mode. The current delay built into the updater application was sufficient for Win10 but not for Win11, so mphidflash was being run too soon, before the EBB was fully recognized by Win11, thus the error.

The solution is very simple, increase the delay before invoking mphidflash. A new updater application will be pushed shortly with this fix.

EmbeddedMan, 2022-05-20:
OK, new EBBUpgraderGUI_v280.exe pushed up. Tested on two Win11 systems and one Win10 system - all work properly.

closed 2022-05-20 by EmbeddedMan