evil-mad/EggBot #183: EggBot Linux Hatch Fill Extension Bug

DrWalrus33, 2022-08-01:
Was trying to hatch fill an object in Inkscape and was met with this error. Any help would be very much appreciated!
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.4 LTS
Release:        20.04
Codename:       focal

```
/usr/local/lib/python3.8/dist-packages/ink_extensions/cspsubdiv.py:12: DeprecationWarning: ffgeom.__init__ -> inkex.transforms.Segment(((x1, y1), (x2, y2)))
  s1 = Segment(p0,p3)
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "eggbot_hatch.py", line 1736, in <module>
    e.affect()
  File "/usr/local/lib/python3.8/dist-packages/ink_extensions/inkex.py", line 291, in affect
    self.effect()
  File "eggbot_hatch.py", line 1215, in effect
    self.recursivelyTraverseSvg([self.selected[id_]], self.docTransform)
  File "eggbot_hatch.py", line 894, in recursivelyTraverseSvg
    self.addPathVertices(simplepath.formatPath(a), node, mat_new)
  File "eggbot_hatch.py", line 730, in addPathVertices
    subdivideCubicPath(sp, float(self.options.tolerance / 100))
  File "eggbot_hatch.py", line 575, in subdivideCubicPath
    if cspsubdiv.maxdist(b) > flat:
  File "/usr/local/lib/python3.8/dist-packages/ink_extensions/cspsubdiv.py", line 12, in maxdist
    s1 = Segment(p0,p3)
  File "/usr/share/inkscape/extensions/inkex/deprecated/meta.py", line 59, in _inner
    return func(*args, **kwargs)
  File "/usr/share/inkscape/extensions/inkex/deprecated-simple/ffgeom.py", line 49, in __init__
    super(Segment, self).__init__((e0, e1))
  File "/usr/share/inkscape/extensions/inkex/transforms.py", line 1098, in __init__
    start, end = other.start, other.end
AttributeError: 'tuple' object has no attribute 'start'
```

oskay, 2022-08-01:
How did you install this?

DrWalrus33, 2022-08-01:
Downloaded the zip and unzipped contents to /root/.config/inkscape/extensions

oskay, 2022-08-01:
Please uninstall those items and try installing them in your user extensions directory instead.

DrWalrus33, 2022-08-01:
Oh okay thanks!

oskay, 2022-08-01:
Please let me know if that works for you. 👍

oskay, 2022-09-26:
I'm going to close this, as it *appears* to be an installation error rather than a new issue with the software. Please feel free to reply if this is not the case.

closed 2022-09-26 by oskay