evil-mad/EggBot #188: USB command processing bug in long commannd strings

EmbeddedMan, 2022-11-25:
When using TeraTerm to send long strings of commands all at once, the EBB will get confused, incorrectly interpret commands (generating errors in parsing), and get into a loop where one of the commands will continue executing forever. The EBB must be reset with the RST button (or powered off) to stop the incorrect motion.

Here is an example of the commands which can trigger this:
```
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
```

Here is the  result

```
SM,1000,1000,1000
OK
SM,1000,1000,1000
OK
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
OK
SM,1000,1000,1000
SM,1000,1000,1000
OK
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
OK
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
OK
OK
OK
SM,1000,1000,1000
SM,1000,1000,1000
SM,1000,1000,1000
OK
OK
OK
!5 Err: Need comma next, found: 'S'
!5 Err: Need comma next, found: 'S'
OK
OK
OK
OK
OK
OK
OK
!5 Err: Need comma next, found: 'S'
!5 Err: Need comma next, found: 'S'
!5 Err: Need comma next, found: 'S'
```

The last section (of OKs and errors) continues forever, and the motion continues forever.

This happens on all versions of EBB firmware (confirmed going back to v2.0.1).

Shorter strings of commands sent all at once do not seem to cause this problem (not sure exactly how much shorter is required to stop triggering the problem).

Commands which take less time (for example, if all the commands in the above string are 100ms long rather than 1000ms long) do not produce the problem.

It feels like the problem would be in the ProcessIO() function, maybe having to do with string array indexes getting corrupted somehow.

It's very surprising to me that this problem has not been seen until now.

oskay, 2022-11-26:
Wow, that is really something. I'm genuinely surprised; I've never run into this either.

EmbeddedMan, 2022-11-26:
Yeah, one of two things could be happening. Either this isn't really a bug in EBB, it's a problem with my hardware or my Windows OS USB host software or TeraTerm or something else not related to EBB. Or it really is a bug in EBB but because of the way EBB has been used to date it's never been triggered before.

oskay, 2022-11-28:
I'm not immediately able to reproduce this, using a python script with that same set of commands.

Script:
```
import sys
from plotink import ebb_serial

def query(port_name, cmd):
    ''' Simple serial query function '''
    port_response = ''
    if port_name is not None and cmd is not None:
        try:
            port_name.write(cmd.encode('ascii'))
            port_response = port_name.readline()
            n_retry_count = 0
            while len(port_response) == 0 and n_retry_count < 100:
                port_response = port_name.readline()
                n_retry_count += 1
        except:
            print("Error reading serial data.")
    return port_response

the_port = ebb_serial.openPort()
if the_port is None:
    print("failed to connect")
    sys.exit() # end script

the_port.reset_input_buffer()

command_list = [
"SM,1000,1000,1000", "SM,1000,1000,1000", "SM,1000,1000,1000",
"SM,1000,1000,1000", "SM,1000,1000,1000", "SM,1000,1000,1000",
"SM,1000,1000,1000", "SM,1000,1000,1000", "SM,1000,1000,1000",
"SM,1000,1000,1000", "SM,1000,1000,1000", "SM,1000,1000,1000",
"SM,1000,1000,1000", "SM,1000,1000,1000", "SM,1000,1000,1000", 
"SM,1000,1000,1000", "SM,1000,1000,1000", "SM,1000,1000,1000",
"SM,1000,1000,1000", "SM,1000,1000,1000",
]

response = query(the_port, 'V\r')
print(response)

for command in command_list:
    print(command + " :: " + str(query(the_port, command + '\r')).strip())

the_port.close()
```

Relevant part of response:
```
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
SM,1000,1000,1000 :: b'OK\r\n'
```

EmbeddedMan, 2022-11-28:
Thank you very much for trying. This is potentially really good news. It could mean that the problem is on the PC side (with my setup, or TeraTerm, etc.) and not in EBB. I will try out some different PCs with different sending systems to try and get to the bottom of where the problem lies.

oskay, 2022-11-28:
As a starting point, maybe try running that Python script yourself.

EmbeddedMan, 2022-11-30:
Your python script runs fine for me - no issues.
I modified it to string together all the commands as one long string with \n and then \r\n between each command - still works just fine.
I then tried copy/pasting the commands above that trigger the problem into RealTerm (another serial terminal emulator) and everything worked fine. So this appears to be an issue with TeraTerm, not with EBB. I am closing the issue.

closed 2022-11-30 by EmbeddedMan

oskay, 2022-11-30:
That's great to hear!