evil-mad/EggBot #190: Add new command to EBB firmware: Set stepper disable timeout

EmbeddedMan, 2023-01-15:
This would be a new command (or an additional part of an existing command like "SC"). It would set a timeout, in milliseconds, which would default at power on to 0.

After the user sets this new timeout to something other than 0 (via the new command), then any time stepper motion stops, the timeout would begin counting down. If it reaches 0, then the EBB will disable the two stepper drivers, meaning the steppers become 'unpowered' (i.e. free to spin on their own). 

During this timeout period, any command which causes the steppers to move would stop the countdown, and when that stepper motion ceased, it would start over, counting down from the set value.

The "QM" command will also have an additional value - it will print out the current value of this countdown timer.

One good way of implementing this is as a countdown timer variable. Any time the variable is not zero, the low_ISR 1ms tick code section would decrement it, and if it then became zero, would disable both steppers. The high_ISR (motor step ISR) would load this variable with the value set by the user after completing a step motion command when there are no more commands in the motion FIFO.

oskay, 2023-01-15:
> After the user sets this new timeout to something other than 0 (via the new command), then any time stepper motion stops, the timeout would begin counting down. If it reaches 0, then the EBB will [...]

It sounds like the possibility exists -- with a short enough timeout or long enough motion command -- for the counter to expire and disable the motors while a motion command is still running.

EmbeddedMan, 2023-01-15:
We would of course have to guard against such a possibility. As outlined above, the countdown timer only gets loaded with a non-zero value at the end of a stepper motor command (when the motion stops) if and only if there are no commands waiting in the FIFO. Any new command that is parsed which causes motor motion not only enables the motor drivers but also sets the countdown timer to zero, thus disabling it if it was in fact counting down at that point.

Instead we could make the disabling of the countdown timer happen in the stepper ISR when a new command is loaded from the FIFO. That would mean that the only place the countdown timer gets loaded is in the stepper ISR, and it is only decremented and checked in the 1ms ISR. That should eliminate any chance of a race condition between the USB command parsing code (which does not happen at ISR time) and the setting/checking of the countdown timer (which would only happen at ISR time).

EmbeddedMan, 2023-09-22:
Version 3.0.0-a17, which has just been merged into the EBF_v3.0.0 branch, now has this feature. The command CU,61,<NewStepperDisableTimeout> controls the feature. Documentation has  been updated to include information on how the feature works.
I did not update QM, but instead added a QU,61 command to retrieve the current value of the StepperDisableTimeout.

closed 2023-09-22 by EmbeddedMan