evil-mad/EggBot #196: DeprecationWarning for Inkscape 1.1 pythont extension

xsnoopy, 2023-02-23:
Hi, using Inkscape 1.1 with the eggbot extension. I recieve the following error. 

``/home/user/.config/inkscape/extensions/eggbot_presethatch.py:24: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`. You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--title")

/home/user/.config/inkscape/extensions/eggbot_presethatch.py:52: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()``

oskay, 2023-02-23:
How did you install the software?

xsnoopy, 2023-02-23:
I just copied the filed in the .config/inkscape/extension folder. I read that in some documentation, but can't find where...

oskay, 2023-02-23:
It sounds like you may have skipped the full install, so you are missing some dependencies.
https://wiki.evilmadscientist.com/Installing_software

oskay, 2023-04-07:
Closing due to inactivity and presumed resolution. Please feel free to comment/reopen if this is not the case.

closed 2023-04-07 by oskay