evil-mad/EggBot #197: Inkscape 1.2.2 not corresponds with EggBot_250A

ralfsoegel, 2023-04-07:
The EggBot_250A can't find the Inkscape installation.
Error message: inkscape not found. Please install Inkscape 0.91 (or newer) and try again.
It is a windows 10 machine.

oskay, 2023-04-07:
Is there a specific reason that you need to run this very old (2015) version rather than the current software?

ralfsoegel, 2023-04-07:
Hi, this version was the direct inscape link on Github. The EggBot_240A askes for the inkscape directory, not the _250A.Hmm...

ralfsoegel, 2023-04-07:
Yes my fault, the inkscape version is like in the title the 1.2.2.2022.

oskay, 2023-04-07:
I really don't follow. Is there a reason that you need to use these very old versions of the EggBot software, rather than the current version?

ralfsoegel, 2023-04-07:
No, this old version is not required, i was a little hopeful to get the Eggbot in motion with smaller software skips. 
The installation of Inkscape 0.92.4 is on the run after that it cames the r281 beside. I hope it will work tomorrow! ;-)

oskay, 2023-04-07:
Since you are using Inkscape 1.2, I would suggest using the current software version, with installers at: https://wiki.evilmadscientist.com/Installing_software

closed 2023-04-07 by oskay