evil-mad/EggBot #198: anti virus trigger

rhawley88, 2023-04-10:
I just downloaded the zip to update my egg bot and Inkscape, and antivirus triggered on this .exe and none of the others. Sorry wasn't sure the best way to bring this to your attention. I'm only on this page 2 or 3 times a year.

EggBot-master\EggBot-master\EBB_firmware\EBBUpdater\deprecated\EBBUpgraderGUI_v280.exe tried to load a malicious resource detected as Gen:Trojan.Heur.RP.MmW@bihjB!ei and was blocked. Your device is safe.

oskay, 2023-04-10:
Unfortunately there isn't anything that we can do about these types of detections in general.

A heuristic detection like that means that this is a file that's an .exe, and that an unknown-to-them .exe file could potentially download something from the internet. (In this case, it doesn't.)  For every different malware protection program/app/database, and for every version of every updater and installer that we manage, there's always some chance for random false positives like these.

I've submitted it as a false positive for that particular company, but that's the only thing that we can do to help. Most companies push updates that allow the after we report false positives, but since they rely on fingerprints, they won't be good for future versions. Note for example that I submitted the *current* release not the deprecated one that you mention.

closed 2023-04-10 by oskay

rhawley88, 2023-04-10:
Thanks yeah I was surprised to see it trigger. I've been downloading from the github for years for my eggbot and never had it trigger before. was more surprised that it only triggered on the one. Figured it was a false positive but wanted it to be brought to your attention. Sorry for taking up your time. Sent from my Galaxy