evil-mad/EggBot #216: EBB Firmware: Comment out code for "PC", "PG" and "T" commands

EmbeddedMan, 2023-09-09:
For v3.0.0, these commands will not be included. The documentation will have the following note for each:

This command is not included in v3.0.0:
Commands PC, PG, T have been marked as "not in use" and tentatively removed in this firmware release.  If your application does use one or more of these commands, please contact us and let us know. If we don't hear from at least a couple of users that these are important, we'll go ahead and remove them permanently in a future firmware version.

oskay, 2023-09-09:
Good. Please leave this issue open to document the change for the time being.

EmbeddedMan, 2023-09-19:
Version 3.0.0-a12 committed to EBF_v3.0.0 branch, with documentation updated and EBB firmware updated as per this issue. Leaving the issue open.

EmbeddedMan, 2024-05-23:
Issue resolved as v3.0.0 now merged into master in PR #232 . This includes updated EBB command documentation indicating that these commands are not present in the firmware HEX file but are still present in the code should they need to be reactivated at some point.

closed 2024-05-23 by EmbeddedMan