evil-mad/EggBot #217: EBB Updater: Only show COM ports that are EBBs

EmbeddedMan, 2023-09-30:
It may be problematic to show (and thus open and try to communicate with) COM ports which are not EBBs. It would be good to filter the COM port list so that only EBBs (as determined by PID/VID combo) are shown and opened.

EmbeddedMan, 2023-09-30:
Also, it would be quite nice if the Windows version of the EBB Updater would be able to handle the situation where the EBB is 'stuck' in bootloader mode (either because of a previous failed update, or the user has put the EBB in bootloader mode, or some other reason). It could do this by doing what the Mac updater does, and if there are no found EBB COM ports, then simply run the firmware update assuming that there is an EBB out there in bootloader mode. Either the update will succeed because there really was an EBB in bootloader mode connected, or it will fail because there is no EBB in bootloader mode.

EmbeddedMan, 2023-09-30:
A new version of the Windows EBB Updater (still for v2.8.1 firmware) has been pushed to the feature/issue_217 branch which has both of the above features. It only looks for actual EBB COM ports (based on USB PID/VID), and if none are found will just try to perform the firmware update assuming an EBB is in bootloader mode.

Here's the link for testing : https://github.com/evil-mad/EggBot/blob/feature/issue_217_EBBUpdaterOnlyEBBCOM/EBB_firmware/EBBUpdater/EBBUpgraderGUI_v281.exe

oskay, 2023-09-30:
https://github.com/evil-mad/EggBot/issues/148

oskay, 2023-10-16:
> Here's the link for testing : https://github.com/evil-mad/EggBot/blob/feature/issue_217_EBBUpdaterOnlyEBBCOM/EBB_firmware/EBBUpdater/EBBUpgraderGUI_v281.exe

I think that there's an issue with this updater.  It fails with:
> Attempting update to version 2.8.1
Incorrect version detected. EBB=  File= 2.8.1 Updated failed.
Click Exit (or click Start Upgrade again to update more EBBs)

EmbeddedMan, 2023-10-16:
Can you describe the test case that created this failure?

oskay, 2023-10-16:
Plugged in an EBB, ran the updater.

EmbeddedMan, 2023-10-16:
What was the exact version of the EBB you plugged in initally?

oskay, 2023-10-16:
It was a 2.9.0, I think. 😅 

Another user reported the same error with version 2.7.0.

EmbeddedMan, 2023-10-16:
Very good. I will test with that combination to reproduce and fix the bug.

EmbeddedMan, 2023-10-17:
I am unable to get the updater to fail on my system using 2.9.0, 2.7.0 and 2.8.1 as starting versions.  I will continue to try different combinations.

In the meantime, are you able to repeatedly get the above linked updater to fail in this way, or was this a one-time thing?

EmbeddedMan, 2023-10-17:
I have now been able to reproduce the bug. I had to use the slowest oldest laptop I have access too in order to do so - I'm not sure if the bug is triggered because the machine is slow, or because it has a bare minimum of drivers (only the latest AD driver has been installed on it, no development tools). But I will now be able to figure out why the bug is happening and hopefully fix it.

EmbeddedMan, 2023-10-17:
I adjusted the timing after the flash update is finished before asking the EBB for it's version string over USB and this appears to fix this bug. Please try out the new installer and see if it now works for you : https://github.com/evil-mad/EggBot/raw/feature/issue_217_EBBUpdaterOnlyEBBCOM/EBB_firmware/EBBUpdater/EBBUpgraderGUI_v281.exe

oskay, 2023-11-17:
Also addresses issue #158

baileysage, 2023-12-21:
Had an issue on Windows 10 with the current released updater where it tried to connect to a Serial over Bluetooth port and then froze and crashed before getting to the EBB's port. 

[image]

[image]

After talking with @EmbeddedMan, he sent me the link to his fix above to test out. Serial over Bluetooth ports successfully skipped, firmware reported as updated on EBB, and Inkscape is no longer giving me "outdated firmware" notices! 

[image]

EmbeddedMan, 2023-12-21:
@oskay This is pretty good proof that the new installer in fact solves the problem that caused this version to be created. So that's fantastic. Thanks so much @baileysage !

oskay, 2023-12-21:
Let's go ahead and bring this into main.

EmbeddedMan, 2023-12-21:
Will do.

EmbeddedMan, 2024-04-03:
These new features for the installer have been merged into the EBF_v3.0.0 branch, is being used for all v3.0.0 and above EBB firmware installers and appear to be working well. When v3.0.0 gets merged into master, it will become the main installer version.

EmbeddedMan, 2024-05-23:
And the v3.0.0 branch has now been merged into master in PR #232 . So the official Windows Updater now has the ability to detect and update EBBs that are already in bootloader mode, and will not open up any non-EBB COM ports when attempting to detect the EBB to update.

closed 2024-05-23 by EmbeddedMan