evil-mad/EggBot #224: Feature request: EBB updater should read and write nickname

oskay, 2024-04-17:
It would be super nice if the EBB updater program could read the EBB nickname (if set), and then write that back to the machine after updating the firmware.

EmbeddedMan, 2024-04-17:
I think this is a fantastic idea, and shouldn't be too hard.

oskay, 2024-04-17:
Awesome. That will solve several independent problems! We'll do the same on the Mac updater.

EmbeddedMan, 2024-04-20:
A new version of the Windows updater is now available on this branch : https://github.com/evil-mad/EggBot/blob/feature/Issue_224_UpdaterNickname/EBB_firmware/EBBUpdater/EBBUpgraderGUI_v301.exe

I need someone other than me to test this before we merge it into the EBF_v3.0.0 branch.

It will retrieve (if possible) the EBB nickname tag before the update, and then set it after the update has completed. It also reads back the set nickname tag to verify that it got set correctly.

I tested it with blank nickname, "x" and "0123456789ABCDEF" and it worked. I also tested it starting from v2.01 firmware (which does not have the "ST" or "QT" commands) and it properly detected that situation.

Note that (at least for me, on fully updated Win10) this new .exe immediately triggered a Microsoft Defender virus alert when I copied it to my desktop. I'm not sure why this version has this effect but previous versions of the updater do not. It's a false positive, and if users have admin access to their PCs they can go into Windows Security and allow this 'threat' so it doesn't trigger Defender in the future. I'm also not sure why this exe is fine in a sub directory of a different drive but copying to my desktop causes the problem.

Also note that because we name EBB Installers by the EBB firmware that they update to, this .exe is named identically to the previous version. So you won't be able to tell by the .exe name that this is a new version. However, this version always prints out "Checking for EBB nickname" in the window as it runs, so you can tell that way.

oskay, 2024-04-20:
Great; thanks. I'm not sure if I'll have access to a Windows machine to test it before Monday. I'm planning to use the same logic on the Mac updater.

oskay, 2024-04-28:
I did not test it yet -- getting there. Is the name readout tolerant of both "future" and "legacy" syntax mode?

EmbeddedMan, 2024-04-28:
Ahhh, good point. No, not currently. I'll fix that and update the .exe. I'll just turn off future mode as the first thing the updater does to the EBB.

EmbeddedMan, 2024-04-28:
OK, new Windows EBB updaters now pushed. It will properly handle an EBB that is in 'future syntax mode'. : https://github.com/evil-mad/EggBot/blob/feature/Issue_224_UpdaterNickname/EBB_firmware/EBBUpdater/EBBUpgraderGUI_v301.exe

oskay, 2024-04-28:
Great; thanks!

oskay, 2024-04-28:
How will older firmware (think ~2.4) handle the command to turn off future syntax?

EmbeddedMan, 2024-04-28:
I tested it with v2.0.1 firmware, and the command generates an error, which
the updater ignores. It works fine.

EmbeddedMan, 2024-04-28:
Just like v2.0.1 doesn't have the QT or ST commands - they generate
'unknown command' errors, which the updater properly ignores (but it
understands that there is no tag if the QT does not generate the proper
response).

oskay, 2024-04-28:
Great; I'm needing to get there as well on the Mac updater.

EmbeddedMan, 2024-05-23:
The EBB v3.0.2 Windows Updater (now merged into master in PR #232 ) has this feature and it appears to work quite well.

closed 2024-05-23 by EmbeddedMan