evil-mad/EggBot #238: Mistakes in EBB S2 command documentation

EmbeddedMan, 2025-02-23:
Kindly submitted by Ilan Moyer:

- The order of parameters is given as S2,Position,Output_Pin[,Rate[,Delay]]<CR>, but in the examples rate and delay are flipped.
- The units of rate are given as 1/12000s per 24ms, but this gives a rate of 3.472ms/s, which seems way too high. Is this supposed to say 1/12000ms per 24ms (3.4722us/S)? That would match the examples later in the section.

EmbeddedMan, 2025-02-23:
The first finding is correct. I am creating a branch with a fix to the S2 documentation which corrects this mistake. I also found another mistake in the third (S2,10000,5,100) example. It should read S2,10000,5,0,100 and is fixed in the branch. I also added the word 'optional' to the Slew Rate parameter description to make it clear that this is an optional parameter. I also moved the Slew Rate parameter description up to before the Rate parameter description so that the order matches the actual parameters in the command.

The second finding is also correct. There were several places in the S2 documentation which used 1/12,000 s as the units for the PWM duration, and other places that got it correct (1/12,000,000 s).

oskay, 2025-02-23:
Agreed that the rate and pin are flipped in the examples. ✅ 

Yes on `1/12,000 of a second` -- That probably was to be `1/12000 of a ms` at some point. 🤦 

So, yes, good catches all around. 👍

closed 2026-10-18 by oskay